Someone-s-out-there/ProjectCarHAN #46: [FEAT] Accu bescherming

GhostJulian, 2023-11-07:
### Is your feature request related to a problem? Please describe.
De [tamiya stekker](https://www.startpage.com/av/proxy-image?piurl=https%3A%2F%2Fasset.conrad.com%2Fmedia10%2Fisa%2F160267%2Fc1%2F-%2Fnl%2F224027_BB_00_FB%2Fimage.jpg&sp=1699345986Tc0d1877151bdd65fa46744ef9ac1b476799f0c1ca86f90cf2d0485576cdb9e41) is redelijk foolproof, maar het kan nog steeds gebeuren dat de polen (+ en - ) verkeerd worden aangesloten.

### Describe the solution you'd like
Zekering  en diodes tussen accu en motordriver monteren zodat de accu en robotauto beschermt is tegen verkeerd aansluiten van accu en te grote/kortsluit stromen.

### Describe alternatives you've considered
Als alternatief kan dit integreert worden in de custom H-brug.

Someone-s-out-there, 2023-11-16:
accu protection has been made

GhostJulian, 2023-11-23:
Door gebruik te maken van de accu bescherming werkt maar 1 moter als beide het stuursignaal krijgen. Door deze module niet te gebruiken heb je dit probleem niet.

We moeten dit onderzoeken waarom dit ontstaat.

GhostJulian, 2023-11-27:
# UPDATE 27-11-2023

Tijdens het meten zag ik dat de spanning met ongeveer 1,6 volt wordt verlaagd. Dit komt door de zener diodes.
Hierdoor wordt de motor driver gevoed met de spanning tussen de 5,6 en 6,4 volt. Dit kan het probleem verklaren.

GhostJulian, 2023-12-11:
# UPDATE 07-12-2023

Accubescherming heeft nu 3 revisies.

### revisie 2 
Introductie van voltage divider om spanning te kunnen meten
Deze heeft 2 normale  9v diodes. Helaas wordt de spanning zo laag dat dit problemen oplevert voor de motordriver

### revisie 3
1 diode weggehaald, accubescherming werkt nu naar behoren.